frangoteam/FUXA #1705: How to let unlogged user see value ?

steve0023, 2025-04-22:
Hi,

Is there a way to let not logged user see editable field, without let them the ability to modify them ?

If I set something in the "enable" column in the authorization of a field. The un logged user cannot see it anymore.
I have to disable the authorization of the field fully to get the value appear but its also editable and that's not what I want.

Or maybe is there a way to set the unlogged user some viewer right ?

Thanks

unocelli, 2025-04-27:
Hi, depends on what kind of control you want to use

unocelli, 2025-05-04:
The feature is now available in versione 1.2.5.
'guest' authentication to show connected View in logout mode

steve0023, 2025-05-05:
Hi, thanks.
It now works with the following setting 
[image]
Guest can read edit field but not change them

closed 2025-05-05 by steve0023