frangoteam/FUXA #1709: Feature Request: Macro Definitions, Tabs, Historical Data, Live Reload

Pencexiang, 2025-04-24:
Feature Request Summary

This issue contains a collection of four feature-related questions or enhancement requests for FUXA. These are important for improving usability and scalability in industrial applications.

1. Macro Definition Support 

Question:
Is there currently any built-in support for macro definitions in FUXA (similar to WinCC or KingView), allowing users to reuse the same UI page with only different variable bindings?

Purpose:
To simplify configuration for devices that share identical UI structure but have different variable paths, reducing the amount of repetitive work.

Expected Enhancement:
Allow page components to accept parameters (e.g., variable prefixes or device IDs) that can be dynamically substituted at runtime.

2. Tab Page Components 

Question:
Does FUXA support tabbed interfaces within a single page, such that multiple views or logical groups can be accessed by clicking tab headers?

Purpose:
To display grouped data (e.g., status, alarms, logs) on one screen without creating multiple separate pages.

Expected Enhancement:
Add built-in support for tab-based UI, possibly using Angular Material `<mat-tab-group>` or similar approach, with JSON configuration.

 3. Historical Data Query 

Question:
Is there a built-in feature for historical data querying within FUXA? Can a user select a variable and view its historical trend over a specific time range?

Purpose:
To allow maintenance or operations teams to trace the behavior of signals over time, without needing to integrate external dashboards.

Expected Enhancement:
Support optional integration with time-series databases (e.g., InfluxDB, TimescaleDB) and provide time-range UI and charting components.

4. Dynamic Reloading / Hot Update 

Question:
Does FUXA support dynamic reloading of runtime pages when changes are made in the Web Editor? Currently, users must manually refresh the browser to see updates.

Purpose:
To improve workflow efficiency during HMI development and deployment.

Expected Enhancement:
Support hot reload of page configurations when edited, either via WebSocket push or client-side polling with auto-reload when changes are detected.

 Summary

These enhancements aim to:
 Improve HMI design reusability (via macro/template support),
 Increase UI flexibility (via tab components),
Enable more powerful diagnostics (via historical queries),
 Speed up development (via dynamic live reload).

Looking forward to the development team’s thoughts. Thank you for this great open-source project!

jingshui127, 2025-04-24:
感觉你根本没有使用过FUXA😊

unocelli, 2025-04-27:
Hi, Regarding your points:
- For Tab Page Components (2) and Dynamic Reloading / Hot Update (4), these are indeed features that would require new development. It would be helpful if you could create a separate issue for each of them, so we can track and prioritize them properly.
- For the other points (Macro Definition Support and Historical Data Query), I recommend checking the existing documentation — which is available across the Wiki and some video tutorials — where you can find useful information related to these topics.

xigua1688, 2025-05-29:
@Pencexiang   hello  nice Pencexiang，could you share your EPICs plugin code , I'm also studying epics. many tks!