frangoteam/FUXA #1714: [BUG]

Rainproof, 2025-05-02:
FUXA version: 1.2.5
Node.js version: v18.19.1
npm version: 10.2.4
Platform/OS: Windows 10 19045.5608
Browser: Google Chrome 135.0.7049.115

start server:
PS D:\SCADA\prjs\ELCADA\FUXA-1.2.5\server> npm start

> fuxa-server@1.2.5-2292 start
> node main.js

2025-05-02T08:55:32.734Z [INF]  FUXA V.1.2.5-2292
2025-05-02T08:55:34.682Z [INF]  FUXA init in  2168ms.
2025-05-02T08:55:34.689Z [INF]  'FUXA Server' created
2025-05-02T08:55:42.490Z [INF]  'opc.tcp://192.168.196.58:4840/' created
2025-05-02T08:55:42.493Z [INF]  'FUXA Server' start
2025-05-02T08:55:42.494Z [INF]  'opc.tcp://192.168.196.58:4840/' start
2025-05-02T08:55:42.497Z [INF]  FUXA started!
2025-05-02T08:55:42.503Z [INF]  WebServer is running http://127.0.0.1:1881/
2025-05-02T08:55:42.518Z [INF]  'FUXA Server' restored 0/0 values
2025-05-02T08:55:42.794Z [INF]  socket.io client connected
2025-05-02T08:55:42.797Z [INF]  socket.io client connected
2025-05-02T08:55:43.015Z [INF]  'opc.tcp://192.168.196.58:4840/' restored 0/0 values
2025-05-02T08:55:43.380Z [INF]  socket.io client connected

start client:
PS D:\SCADA\prjs\ELCADA\FUXA-1.2.5\client> npm start  

> fuxa@1.2.5-2292 start
> ng serve

✔ Browser application bundle generation complete.

Initial Chunk Files   | Names         |  Raw Size
vendor.js             | vendor        |  15.92 MB |
main.js               | main          |   3.27 MB |
styles.css, styles.js | styles        | 553.90 kB |
polyfills.js          | polyfills     | 316.70 kB |
scripts.js            | scripts       |  42.18 kB |
runtime.js            | runtime       |   8.34 kB |

                      | Initial Total |  20.09 MB

Build at: 2025-05-02T09:07:25.099Z - Hash: 6d0834ca6248e858 - Time: 23515ms

Warning: D:/SCADA/prjs/ELCADA/FUXA-1.2.5/client/src/environments/environment.client.ts is part of the TypeScript compilation but it's unused.
Add only entry points to the 'files' or 'include' properties in your tsconfig.

Warning: D:/SCADA/prjs/ELCADA/FUXA-1.2.5/client/src/environments/environment.demo.ts is part of the TypeScript compilation but it's unused.
Add only entry points to the 'files' or 'include' properties in your tsconfig.

Warning: D:/SCADA/prjs/ELCADA/FUXA-1.2.5/client/src/environments/environment.prod.ts is part of the TypeScript compilation but it's unused.
Add only entry points to the 'files' or 'include' properties in your tsconfig.

Warning: D:\SCADA\prjs\ELCADA\FUXA-1.2.5\client\src\app\maps\maps-view\maps-view.component.ts depends on 'leaflet'. CommonJS or AMD dependencies can cause optimization bailouts.
For more info see: https://angular.io/guide/build#configuring-commonjs-dependencies

** Angular Live Development Server is listening on localhost:4200, open your browser on http://localhost:4200/ **

√ Compiled successfully.

The page http://localhost:4200/ does not open.
A message appears "Server connection failed!"

[image]

unocelli, 2025-05-04:
Could you please open the browser DevTools (F12), go to the **Network** tab, and then refresh the page at [http://localhost:4200](http://localhost:4200)?  
Look for any failed requests — especially those pointing to `http://localhost:1881`, `http://localhost:1881/api/project` or showing status codes like `404`, `0`, or `ERR_CONNECTION_REFUSED`.

Also, check the **Console** tab for any error messages.

carlbomsdata, 2025-05-04:
I have noticed that as well. If you inspect the page i guess that you will see some socket.io errors.

At least for me, the quickfix is to build the frontend (npm run build)

Kiruthick-Roxx, 2025-05-07:
Hi,
I was also facing this issue since the last release.
I forgot to raise an issue.

Kiruthick-Roxx, 2025-07-05:
Hi @unocelli,

Actually, this problem has begun after the commit (23cac9e [fix: cors security with settings in settings.js](https://github.com/frangoteam/FUXA/commit/23cac9e5e5130478d865d7f438767141c6a30060) ).

It is blocking localhost:4200 from accessing the server resources.