frangoteam/FUXA #1715: [QUESTION] Reverse Proxy when working with multiple fuxa instances

AndreCadena, 2025-05-02:
My team and I have created an NGINX reverse proxy to redirect any access to a single localhost port (like http://localhost:1881). The problem is, we tried to move the loading of the file from the seed to /Client1/, in order to have https://{domain_name}/Client1/ redirect the user to http://localhost:1881 and the same thing with /Client2/ and port 1882.

We have succeded to do it when running a single fuxa instance, for example if there is only the Client1 and port 1881. When establishing multiple containers in our vps server or in a local machine, the frontend part can be correctly connected, meaning the redirection works, but the Client2 fuxa instance is trying to use Client1's files (socket.io, app and assets). The intended use is for each fuxa instance to use their own separate files, and they should never get in eachother's way. Can anyone help us with a bit of guidance into how can one setup the reverse proxy to correctly redirect each fuxa instance to their own files?

We have been working with docker using the latest image available by frangoteam, building newer versions of the image as we try to solve the issue. We are all pretty unfamiliar with the networking side of docker applications like this so a little help would be greately apreciate.

FUXA is an amazing SCADA system thank you all for your work.

MatthewReed303, 2025-05-02:
@AndreCadena I have done similar but I run all my Fuxa instances in Docker Compose and change port in compose file, this way they all use there own resources. I also use Nginx Proxy Manager.

AndreCadena, 2025-05-05:
It is possible our team has messed up the NGINX config file. Is it possible you could send us the NGINX config parameters you used? I fail to see how setting up a single compose file for all fuxa instances changes from setting up a different compose file for each fuxa. In theory the parameters sent to docker compose are the same, be that environment variables, ports, separate volumes or docker networks.

H-hejin, 2025-06-27:
Has there been any progress on this issue

envas, 2025-10-14:
> Have you looked into this?

No and I'm afraid my skills aren't up to the task. If it were written in Java, maybe, but all that Node/JavaScript stuff with a million libraries that I have no idea what they actually do is nothing for me.