frangoteam/FUXA #1728: [BUG] Lost connection and reconnect issue

carlbomsdata, 2025-05-04:
Hi,

I just noticed something if a client (browser connected to fuxa) loses network connection to fuxa.
When client loses network connection, after some time a message shows Server connection failed (which seem correct), but while in this state, if i click to set tag values nothing happens in the ui (which is correct since we are not connected to server).
But when client reconnects to server, these tag value changes was indeed sent to server opon reconnection, BUT it is not showing it the ui until client refresh browser. 

This does not seem correct. Or am i missing something? 

I tried recoding this issue here:

https://streamable.com/2d0818

I run fuxa version 1.2.4-2188

carlbomsdata, 2025-05-04:
As I see it a quick fix would be show the red banner with 100vh open disconnection, this would disable user input whilst disconnected.
Also decreasing the timeout for when banner gets shown. For now, after disconnect, it takes around 5 seconds until the banner gets shown.
Sadly I cannot find where this timeout is set in the code.

unocelli, 2025-05-07:
Hi, good idea, I suggest to create a panel just below the banner that spans the full width of the page and in black with opacity

in home component
<div *ngIf="serverErrorBanner$ | async" class="banner">
	<div>
		<span>{{'msg.server-connection-failed' | translate}}</span>
		<mat-icon>block</mat-icon>
	</div>
</div>

closed 2026-11-08 by unocelli