frangoteam/FUXA #1734: [BUG] Unable to communicate with S7?

jingshui127, 2025-05-07:
**Describe the bug**
I have installed the communication plugin for S7, but I still cannot communicate normally!

[image]

[image]

[image]

[image]

[image]
It is normal to use other testing software.

[image]

**Environment**
 - FUXA version:

[image]

 - Node.js version:
 - npm version:

[image]

 - Platform/OS:

[image]

 - Browser:

[image]

jingshui127, 2025-05-07:
[image]

jingshui127, 2025-05-07:
[image]

[image]

[image]

[image]

jingshui127, 2025-05-07:
[image]

tobias-carlbom, 2025-05-10:
Hi,

I have successfully been communicating with S7 300 recently using same version of fuxa as you (and npm package), running on windows 11 and windows server 2019.

At least for me i had to tinker with the rack and slot.

To eliminate config in error and/or fuxa i would advise you to setup a very simple nodejs application using node-snap7. Example can be found here: https://www.npmjs.com/package/node-snap7

jingshui127, 2025-05-12:
> Hi,
> 
> I have successfully been communicating with S7 300 recently using same version of fuxa as you (and npm package), running on windows 11 and windows server 2019.
> 
> At least for me i had to tinker with the rack and slot. For be the Rack=0 and Slot=2.
> 
> To eliminate error in config and/or fuxa i would advise you to setup a very simple nodejs application using node-snap7 to your plc. Example can be found here: https://www.npmjs.com/package/node-snap7

Thank you for your answer. The problem was solved by running it directly.

But after being packaged into an electron, S7 cannot be used normally.
I tested the latest version 1.2.5.
And it has been tested on both WIN10/11 and Windows Server 2022.

closed 2025-07-24 by unocelli