frangoteam/FUXA #1737: [FEATURE] Please clarify how 'Format digits' work

andergisomon, 2025-05-09:
**Describe the feature**
Spent days scouring the issues page until I stumbled upon a comment.
Apparently to show at most 2 decimal points, I have to put 2 into the Format Digits field. This is extremely cryptic.

**Describe the solution you'd like**
Either add a help popup that users can hover near each of these fields or explicitly mention it in the Wiki and other documentation.

**Additional context**

[image]

tobias-carlbom, 2025-05-10:
Sorry to hear that! You are more then welcome to contribute to the project though.
@unocelli is doing quite some heavy lifting on this project and i think he needs every help he can get!

unocelli, 2025-05-18:
@andergisomon sorry I was a bit too prompt, putting a small tooltip is not a bad idea