frangoteam/FUXA #1751: [FEATURE]

RangeyRover, 2025-05-19:
**Additional context**
This is for a rail signalling environment for a signal. Typically this would show:
Grey: No Comms
Red: Signal at Stop
Green: Signal at clear
Flashing Red: Signal timing off

**Describe the feature**
Multiple tags possible to control the status of shapes. I've tried, but do not think it is possible to have multiple tags control the status of a shape. For the context above, I can have Red from the properties tab, and can have flashing red with the actions tab. I cannot see a way to control a single object to other states with different tags. Recognising I can have multiple values in a single tag, but I'd rather avoid that if possible.

**Describe the solution you'd like**
Single shape, multiple tags in the property tab with the logic for controlling the state of the shape elsewhere, with potentially a priority order for obeying the tags.

It's also possible ive got this wrong and this feature already exists.

MatthewReed303, 2025-05-19:
@RangeyRover use an INT to set the different states:

0: Grey: No Comms
1: Red: Signal at Stop
2: Green: Signal at clear
3: Flashing Red: Signal timing off

RangeyRover, 2025-05-19:
Thank you. I saw that as a possibility. In my case that would mean pre sorting my existing message stream from its function based grouping to object based grouping. It is possible.
I would like to have more configurable control from a variety of sources for an object though. 
I think my request is closely related to [https://github.com/frangoteam/FUXA/issues/954](https://github.com/frangoteam/FUXA/issues/954).
In my typical environment of Rail Signalling 1-1 mapping of object state to tag is more typical, ie many tags per object. I understand this may not be the intended direction of FUXA so feel free to close if it is not.

MatthewReed303, 2025-05-19:
why not write a script and convert your bool tag states to an INT tag? 

In the script you would read the bool tags and then assign the value to the INT, you would need some priority logic just in case you ever have the case of multiple bools being true at the same time as you can only display and single state at a time...

https://github.com/frangoteam/FUXA/wiki/HowTo-configure-Script

RangeyRover, 2025-05-20:
Thanks for the response that's a really interesting concept. I'll investigate some more.

closed 2025-07-31 by unocelli