frangoteam/FUXA #1761: [BUG] Ethernet/IP

esdcgit, 2025-05-21:
When using the Ethernet/IP drivers to communicate with a Micrologix 1400 using ST10:0 as a reference with a value of 1234567890 I receive in return 2143658709

I dont know how to fix this.

tobias-carlbom, 2025-05-21:
Hi!

It seems to be a byte swap issue. 

Can you please verify if your controller uses Little Endian or Big Endian?

esdcgit, 2025-05-21:
I have tested with int and bits with no issues so far.  Only on strings.

I dont know how to validate it, but based of the information i can find online it is big endian

tobias-carlbom, 2025-05-21:
Thanks for the information!

A quick fix might be to parse the string in a fuxa server side script (that runs every second).
Which parses your string tag and outputs the result to another tag.

It's not pretty but it will work.

The long term solution would be to implement byte and/or word swap in Fuxa backend and frontend.

tobias-carlbom, 2025-05-21:
I think this parser logic will work but. I am not at a computer right now to adjust this script to fuxa (Read and write to tag).
'''
function fixSwappedString(str) {
  let fixed = '';
  for (let i = 0; i < str.length - 1; i += 2) {
    fixed += str[i + 1] + str[i];
  }
  if (str.length % 2 !== 0) {
    fixed += str[str.length - 1];
  }
  return fixed;
}

'''

esdcgit, 2025-05-22:
I still need to learn how to utilize these function within the Fuxa ui.  But getting there.  Will try and once i have success will let you know

unocelli, 2025-10-03:
I assume the issue has been resolved in the meanwhile.