frangoteam/FUXA #1764: "Virtual-Keyboard full screen" get stuck sometimes

Lampjevp, 2025-05-23:
First of all i want to thank the team and community, i have been FUXA for a bit now and i really like the "product".

I have been experiencing an issue with the keyboard getting stuck in place, for now the only fix i could vind is refreshing the page.
My guess is that there is something happening when you switch between pages while the keyboard is opened, but i'm not sure

**To Reproduce**
I myself have not been able to reproduce the issue

**Environment**
 - FUXA version: 1.2.5
 - Node.js version: v18.20.8
 - npm version: 10.8.2
 - Platform/OS: Raspberry pi OS
 - Browser: Chromium 135.0.7049.84 built on Debian GNU/Linux 12 (bookworm)

unocelli, 2025-05-25:
Hi, Thanks.
A bug is something we can confirm only if it's reproducible.
If you find a way to trigger it consistently, please share the steps.

unocelli, 2025-10-03:
I assume the issue has been resolved in the meanwhile.

closed 2025-10-03 by unocelli