frangoteam/FUXA #1766: Add width and height adjustment to the property bar

levin-go, 2025-05-23:
Sorry, I'm not very familiar with this software yet. 

When I was adjusting the size of the circle, I found that it could only be adjusted by dragging with the mouse, but after the adjustment was completed, I found that it had deformed (not a qualified circle). So it is recommended to add a width and height adjustment function in the property bar.

Or if there are any other good ways, please let me know. 

Thank you!

levin-go, 2025-05-23:
[image]

Similar to this type of control, when I resized it, its aspect ratio became incorrect. Especially when I make it spin again, it feels very awkward

malcore10, 2025-05-23:
[wiki](https://github.com/frangoteam/FUXA/wiki/Tips-and-Tricks)

levin-go, 2025-05-26:
> [wiki](https://github.com/frangoteam/FUXA/wiki/Tips-and-Tricks)

Shift + 'resize with mouse the selected item': lock width and height

The above command solved my problem, thank you！

closed 2025-05-26 by levin-go