frangoteam/FUXA #1768: [FEATURE] Use a non root user in the docker image

Dowster64, 2025-05-24:
**Describe the feature**
The docker image (and the container as run by the compose.yml) use the root user which does not appear to be Docker best practice due to security concerns

**Describe the solution you'd like**
Non root user used in the docker distribution and compose file example.

**Additional context**
If applicable, add screenshots to help explain your idea.

unocelli, 2025-06-26:
Hi,
thanks for raising this. Could I ask if you’ve already tried running FUXA as a non-root user in Docker? Have you encountered any issues during setup or operation?

suttontools, 2025-07-01:
@unocelli I haven't yet.  I've used Alpine in the past as a base image and it requires adding the sudo package (and other packages to install the desired software), creating the new user (usually GUID 1000), then removing the sudo package and making sure the right user starts the app.  I will have a go - not sure if the slim version of your base image includes sudo.  Not sure what what requirements dos2unix needs though.