frangoteam/FUXA #1800: [BUG] Issue with Boolean Value Synchronization via MQTT

GitBaer, 2025-06-08:
I am reading a Boolean value from ioBroker into FUXA via MQTT. When I change the value in ioBroker, the display in the "Connection settings" of FUXA updates correctly (value "true" or "false").

However, when I use this data point in the visualization, it does not reflect the changed value accordingly. The other way around works: if I change the data point in the visualization, the value is correctly transmitted to ioBroker.

From my point of view, this appears to be a more serious issue. I noticed a similar behavior when testing with Boolean variables that are created directly on the "FUXA Server."

[image]

unocelli, 2025-06-26:
Hi, 
This issue is likely related to how FUXA handles MQTT values. Since MQTT topics don’t include the data type, FUXA tries to convert the values to numbers to simplify control logic. But in cases like Boolean values ("true"/"false"), this doesn’t always work as expected in the visualization.

Also, just to better understand your setup: when you change the value from the visualization and it updates in ioBroker, are you using a separate topic for publishing the value? Usually, MQTT subscriptions only receive data, so we’d like to confirm how the write-back is working in your case.

unocelli, 2025-07-24:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2025-07-24 by unocelli