frangoteam/FUXA #1827: I have a Vue or HTML application, and I want to display a specific view of FUXA separately. How can I do it? I have already purchased the Pro version; I am very confused as there is no mention of deployment in the document!

H-hejin, 2025-06-24:
I have a Vue or HTML application, and I want to display a specific view of FUXA separately. How can I do it? I have already purchased the Pro version; I am very confused as there is no mention of deployment in the document!

jimbdooley, 2025-06-24:
I’m still new to Fuxa, so other people can probably give a better answer.

But what I’d look into is, duplicate your project, set the view of interest to be the main view, then in your vue/html project use an iframe and point it to that. I just tested it locally it appears to work.

 Note it does require 2 separate servers. Your websites server + the fuxa server

manish-kumar-developer, 2025-06-25:
its not clear although what you want to do, but you can inject js files and in dom you can create an element and so what you want to show in that, in order to interact with fux you will need to run server, or u can use dist(public) content if you want to show same static part.