frangoteam/FUXA #1871: [FEATURE]Label Mapping

Krita-king, 2025-07-19:
I don't know much about fuxa，But I like fuxa,Can label mapping be achieved? My goal is to map the modbus tcp label to the enthernet protocol label, so that it will be very flexible in multi-device control. My English is a bit bad, I hope you can understand it, thank you for your open source

tobias-carlbom, 2025-07-26:
Hi @Krita-king 

I would appreciate if you could elaborate on what you are trying to achieve.

If you need to "map" eg a modbus adress/register to a Ethernet IP/ip adress is would create a server side script.

You can read more about server side scripts in the Wiki.

Krita-king, 2025-07-30:
Our company is using exor JMobile and finds your open-source scada very similar. Our colleagues think your open-source fuxa is excellent. Everyone likes it, and it doesn't require installing additional development software. However, it's mature commercial software with comprehensive functionality. You can refer to it for iterative upgrades. https://www.exorint.com/ I will learn scripting to better use this system. This is a very good project, and more and more companies will use it. I hope you can add some scripting tutorials on the wiki to help everyone familiarize themselves with it.

tobias-carlbom, 2025-08-03:
@Krita-king All cudos goes to @unocelli Perhaps you two should connect - info@frangoteam.org

Regarding the scripts, please see: https://github.com/frangoteam/FUXA/wiki/HowTo-configure-Script

Scripting in Fuxa is just regular (modern) Javascript. I used https://www.w3schools.com/js/ and youtube when i started learning Javascript 10 years ago :)

Happy hacking!