frangoteam/FUXA #1874: [BUG] Script Scheduled at 00:00 Resets at 12:00 PM Instead

Jue07, 2025-07-21:
Description:
A script scheduled using FUXA's scheduler to run daily at 00:00 (midnight) is instead being triggered/reset at 12:00 PM. This behavior contradicts the defined schedule and affects data accuracy for daily resets or calculations.

[image]

Steps to Reproduce:

1. Open FUXA.
3. Create a script with a scheduled time set to 0:0.
5. Observe that the script does not execute at midnight as expected.
7. Instead, it appears to run or reset at 12:00 PM.

Expected Behavior:
The script should execute/reset at 00:00 (midnight) based on the defined schedule.

Actual Behavior:
The script executes or resets at 12:00 PM instead of midnight.

Impact:
This causes incorrect daily logging and calculations, especially for energy counters and cumulative daily data.

**Environment**
 - FUXA version: 1.2.7 -2426
 - Node.js version: 18.20
 - Browser: Chrome

unocelli, 2025-08-07:
Hi, Thanks for the report!

unocelli, 2025-08-08:
@Jue07 Should be fixed

closed 2025-10-02 by unocelli