frangoteam/FUXA #1889: [BUG]I have more than ten sites, can I view them? Why is all the project information returned when accessing by name? Loading one view would consume a lot of time for me!

H-hejin, 2025-08-01:
I have more than ten sites, can I view them? Why is all the project information returned when accessing by name? Loading one view would consume a lot of time for me!

H-hejin, 2025-08-01:
I have many views, but I only want to access one view, which loads very slowly. It seems that I have loaded all the resources of other views in the interface for obtaining project details

tobias-carlbom, 2025-08-08:
Hi @H-hejin 
You need to explain and show what you mean with "slow" and we also need more context to be able to help you.

What browser? Hardware spec? Software spec? How many tags? Scripts?

Also, can you replicate the "behavior" with a new/fresh Fuxa project?

H-hejin, 2025-08-11:
I only accessed one view, but from the network request, it appears that resources for other views have also been loaded

H-hejin, 2025-12-18:
@unocelli Regarding this matter, do you have any plans to address it

tobias-carlbom, 2026-01-14:
Hi,

I can try and reproduce the issue. But can you please provide an example project and tell me about your setup?

H-hejin, 2026-01-28:
@tobias-carlbom This is the API: https://scada.hatchsolution.cn/api/project. Based on the results returned from this interface, it seems that the data from other attempts I created has also been returned
This is the project address: https://scada.hatchsolution.cn/view?name=二教机组

Actually, I only need the project data for the second-generation teaching unit

H-hejin, 2026-01-28:
Has the Pro version addressed this issue? I have already purchased the Pro version to show my support

tobias-carlbom, 2026-02-03:
I guess we have to make a poll on this. Seems to be significant work to implement lazy-load to app.module.ts and also (perhaps) all components.

Just to get things clear. Is loading all assets a problem when visiting a Fuxa page on LAN only? Atleast for me, loading all assets has never been a problem on LAN only.

H-hejin, 2026-02-03:
@tobias-carlbom We are accessible through the public internet

H-hejin, 2026-02-10:
@tobias-carlbom @unocelli I optimized /api/project through AI; from the actual effect, it indeed only increased the speed; the author can refer to it and synchronize it to the PRO version, as I need to use the PRO version for my work