frangoteam/FUXA #1901: [BUG] Reading M area — works for first 19 tags, but all become empty starting from 20th tag

fcy9697-hue, 2025-08-13:
Hello,

When reading memory area (M) tags from PLC, FUXA can only successfully read the first 19 tags.
If I add the 20th M tag, all M area values become empty.

In the backend logs, I see repeated warnings  like:
[WAR]  'M' working (connection || polling) overload! 1
[WAR]  'M' working (connection || polling) overload! 2
[WAR]  'M' working (connection || polling) overload! 3

[image]

[image]

Krita-king, 2025-09-06:
rtu？

fengchenyu, 2025-09-25:
> rtu？

snap7-1500