frangoteam/FUXA #1912: [BUG] There is a bug in the pop-up page of the device

viaya123, 2025-08-26:
There is a bug in the pop-up page of the device

When I click the search button on this pop-up page, and then navigate to other pages opened in the browser, after waiting for a while and returning, the page becomes blank. After a while, the page recovers, and then there is an error reported in the console。

[image]

[image]

Additionally, I would like to inquire whether the canvas has an adaptive function？

unocelli, 2025-08-27:
Thanks for the report, The error is not ideal, but it should not compromise the functionality.

viaya123, 2025-09-01:
Okay, can we add canvas adaptation later on？