frangoteam/FUXA #1924: [BUG]The Header Navigation does not work when on a Map View

hesitatenot, 2025-09-05:
**Describe the bug**
The Header Navigation does not work when on a Map View

**To Reproduce**
Steps to reproduce the behaviour:
1. Create several Views, including a Map View, and create corresponding Header Navigation Buttons as indicated in https://youtu.be/CgxWorE0Fkc?si=_FNbUQDNyHN6qmrZ
2. Go to Home, navigate to the Map view
3. Click on the other Header Navigation Buttons and nothing happens

**Expected behaviour**
Header Navigation Buttons allow navigation from the Map view to other views

**Environment**
 - FUXA version: Docker image snap7-1.2.6
 - Node.js version: Docker image snap7-1.2.6 
 - npm version: Docker image snap7-1.2.6
 - Platform/OS: Docker image snap7-1.2.6
 - Browser: Chrome

hesitatenot, 2025-09-10:
Hi @tobias-carlbom 

I've attached the debug project for your review.

As we can see, with the sidebar navigation, we can navigate between MainView, MapView, View1 and View2.

With the header navigation, we can navigate between other views; however, once entered in the MapView, we will not be able to navigate with the header navigation.

Thank you for your support!

[fuxa-debug-project.json](https://github.com/user-attachments/files/22255662/fuxa-debug-project.json)

tobias-carlbom, 2025-09-10:
Hi @hesitatenot 

Many thanks for sharing your project! 

It appears to be a known bug and we will try to fix it as soon as possible.

Thanks for reporting this :)